assembly12/Foxess-T-series-ESPHome-Home-Assistant #1: Fox Ess T10 G3 has another protocolö

4ndinger, 2022-10-14:
Hi there!

First thanks for sharing this project regarding Fox Ess informations been send on RS485. Sadly my Fox Ess T10 G3 has some different frames which are being sent. Did you reverse engineered yours or is there an Frame definition from Fox Ess available?

In my case it sends three different frames with different length. I receive 0x01, 0x02 and 0x06 on byte[2]. But my frame 0x02 is 245 bytes long. Yours seem to be shorter. 

If needed I can send you those three strings!

Thanks and best regards
Andi

assembly12, 2022-10-15:
Hi Andi,

Thanks for your feedback!
I've updated the foxess_t_series.h. It should now work with any bytes length.
Please try it out and let me know if the issue is resolved!

4ndinger, 2022-10-15:
Hey,

thanks I will try it later! Do you have a frame definition available or just reverse engineered all of the informations? I’m interested in the faults and the other data which is sent, too. 

Best regards
Andi

niqu1982, 2022-10-24:
Heare You are spec. from foxess poland
https://drive.google.com/file/d/1eZSxXRflruOIlPNNiN7fuOvuC0YZZzjg/view?usp=sharing

closed 2022-11-27 by assembly12