assembly12/Foxess-T-series-ESPHome-Home-Assistant #3: SenWhich sensor is responsible for the mode of work of inverter?

DiegoFoxy, 2022-11-04:
Which sensor is responsible for the mode of work of inverter?
- Normal 
- Alert
-...

assembly12, 2022-11-04:
As far as I can tell the operating mode isn't broadcasted as such.
However:
When the inverter is offline, there are no comms anyway. So you could check whether the last communication was more than 2 minutes ago; if so the inverter is probably offline.

Registers 52 to 59 contain fault messages. You can add them to the .h file the same way the other sensors are setup. However my inverter hasn't given a single fault ever up till now, so I decided not to include them in the code by default.

DiegoFoxy, 2022-11-14:
I am just learning to program :(. Could you please suggest what the lines of code in the .h file should look like, for registers 52 to 59? I've had two errors during the month, so I'll be a good tester of that.

assembly12, 2022-11-16:
Hi,
for what it's worth: I think your suggestion is very good and I do intend to include it (when I find the time).

For now you could add something like this:

`        uint32_t FaultMessage1_value = int(
            (unsigned char)(bytes[52]) << 24 |
            (unsigned char)(bytes[53]) << 16 |
            (unsigned char)(bytes[54]) << 8 |
            (unsigned char)(bytes[55]));
        ESP_LOGI("custom", "Fault message 1: %i", FaultMessage1_value.UInt16);
        delay(100);`

`        uint32_t FaultMessage2_value = int(
            (unsigned char)(bytes[56]) << 24 |
            (unsigned char)(bytes[57]) << 16 |
            (unsigned char)(bytes[58]) << 8 |
            (unsigned char)(bytes[59]));
        ESP_LOGI("custom", "Fault message 2: %i", FaultMessage2_value.UInt16);
        delay(100);`

`        uint32_t FaultMessage3_value = int(
            (unsigned char)(bytes[60]) << 24 |
            (unsigned char)(bytes[61]) << 16 |
            (unsigned char)(bytes[62]) << 8 |
            (unsigned char)(bytes[63]));
        ESP_LOGI("custom", "Fault message 3: %i", FaultMessage3_value.UInt16);
        delay(100);`

`        uint32_t FaultMessage4_value = int(
            (unsigned char)(bytes[64]) << 24 |
            (unsigned char)(bytes[65]) << 16 |
            (unsigned char)(bytes[66]) << 8 |
            (unsigned char)(bytes[67]));
        ESP_LOGI("custom", "Fault message 4: %i", FaultMessage4_value.UInt16);
        delay(100);`

This is completely untested, so it might need some adjusting. No HA sensors are created for the fault messages, they will however be displayed in the log.

assembly12, 2022-11-30:
As of now the state of the inverter is published in an esphome sensor that can be either "online" "offline" or "error".
If the state is "error", error messages are displayed in the logs.