assembly12/Foxess-T-series-ESPHome-Home-Assistant #5: Failed to prepare configuration

PiotrRowinski, 2022-11-22:
Hi!

I'm not able to prepare the configuration when trying to include foxess_t_series.h:

> INFO Reading configuration /config/esphome/esp32-foxess.yaml...
> INFO Generating C++ source...
> INFO Compiling app...
> Processing esp32-foxess (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
> --------------------------------------------------------------------------------
> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> Dependency Graph
> |-- AsyncTCP-esphome @ 1.2.2
> |-- WiFi @ 1.0
> |-- FS @ 1.0
> |-- Update @ 1.0
> |-- ESPAsyncWebServer-esphome @ 2.1.0
> |   |-- AsyncTCP-esphome @ 1.2.2
> |-- DNSServer @ 1.1.0
> |-- ESPmDNS @ 1.0
> |-- noise-c @ 0.1.4
> |   |-- libsodium @ 1.10018.1
> Compiling /data/esp32-foxess/.pioenvs/esp32-foxess/src/main.cpp.o
> In file included from src/main.cpp:22:0:
> src/foxess_t_series.h:3:63: error: expected class-name before ',' token
>  class foxesscomponent : public PollingComponent, public Sensor, public UARTDevice {
>                                                                ^
> src/foxess_t_series.h:3:83: error: expected class-name before '{' token
>  class foxesscomponent : public PollingComponent, public Sensor, public UARTDevice {
>                                                                                    ^
> src/foxess_t_series.h:5:35: error: expected ')' before '*' token
>      foxesscomponent(UARTComponent *parent) : PollingComponent(600), UARTDevice(parent) {}
>                                    ^

Any ideas?

Thanks in advance!

assembly12, 2022-11-27:
Are you sure you didn't change anything, like renaming the component?
Just try using the Github files without changing anything.

assembly12, 2022-11-30:
What model inverter do you have?
Those serial messages look nothing like what a Foxess T-Series would transmit.

PiotrRowinski, 2022-11-30:
It is a Foxess T5 PV Grid-tied inverter - [https://static.fox-ess.pro/upload/T_G2_PL_f9fb5ece7a.pdf](url). Maybe some additional rs-485 configuration is required on the inverter side?

BTW would You be able to share examples of expected serial messages?

assembly12, 2022-12-01:
This is what a typical output looks like:
`7E 7E 02 63 23 ED AC 00 E8 00 00 17 E8 00 00 09 5B 00 55 13 87 07 F1 09 53 00 55 13 87 07 EA 09 63 00 55 13 87 07 F8 14 DA 00 60 00 00 0E 7F 00 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2C 00 2C 00 2C 00 82 00 00 55 5A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 03 50 78 00 10 00 00 00 00 00 1A 00 00 00 00 00 00 00 00 00 00 02 21 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 02 00 00 00 00 03 E7 FF 64 17 EA 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 3F E5 00 00 00 00 00 00 00 00 00 00 00 00 C6 63 E7 E7`

It always starts with 7E 7E and ends with E7 E7. Most of the Info is in the beginning of the message. In the middle there are a lot of zeros. Total length is either 165 or 245 bytes, depending on the firmware installed on the inverter.

PiotrRowinski, 2022-12-12:
After changing the CT/Meter setting (from _Disable_ to _Comm_), I started getting similar serial messages (with 7E 7E at the beginning), and everything is working fine now!

Thanks for your support!

closed 2022-12-12 by PiotrRowinski