assembly12/Foxess-T-series-ESPHome-Home-Assistant #6: Using LAN adapter instead of RS

nikowitt, 2022-11-28:
Hi there,
in https://github.com/StealthChesnut/HA-FoxESS-Modbus, it is mentioned that the LAN adapter will also work to get the data. Does this also work in this case?
Regards,
Niko

assembly12, 2022-11-30:
Yes, it's possible. It will however not be compatible with this github repository.
[](https://community.home-assistant.io/t/foxess-inverter-modbus/459313/48)

nikowitt, 2022-12-01:
I see, need to have a closer look. Are there place on your end to make this possible for WLAN, so basically providing the capabilities like in https://github.com/StealthChesnut/HA-FoxESS-Modbus for the T series?

assembly12, 2022-12-01:
Hello!
Im not sure if this is a misunderstanding on my part.
In your opening post you asked about LAN and in your recent post you ask about WLAN; is there a typo or autocorrection?

All the code in this component runs on the ESP32, which in turn connects to HA via WLAN (wifi).
Is there a specific reason you don't want to use an ESP-based solution?
There is some soldering involved, but apart from that it's really just copy paste.

nikowitt, 2022-12-01:
I am sorry, "WLAN" was a mistake. I am talking about the LAN adapter as everything is suppose to go over a wired connection. As I am at the beginning of the journey, I am not really familiar with the connectivity of devices yet. From the other repo, I was under the impression that in general, using LAN works without additional hardware but with maybe only some different mapping required.

So in a nutshell, I am looking for an easy way to connect a T series inverter via LAN adapter to HA to add as little additional hardware (and complexity to the whole system) as possible :-)

closed 2022-12-21 by assembly12