assembly12/Foxess-T-series-ESPHome-Home-Assistant #7: Communication stopped working

tomaszsluszniak, 2022-12-21:
Hi, 
I have a problem with communication (I guess) between ESP32 and Inverter (T10G3). It worked for a month or two but after a week of having the solar panels fully covered by snow (the inverter was not working), the snow melted and now it doesn't work anymore. I cannot receive any parameters from the inverter. I'm now sure, but I think that the inverter firmware version had changed. The current version is 1.18, previously it was 1.13 I guess. Or maybe I'm wrong.

I'm able to connect to it (web_server module) but in the logs, I can see only this sequence:

```
13:11:40	[D]	[sensor:127]	
'T-Series Inverter State': Sending state 0.00000  with 0 decimals of accuracy
13:11:40	[D]	[text_sensor:067]	
'T-Series Inverter Mode': Sending state 'Offline'
13:11:40	[D]	[sensor:127]	
'T-Series Inverter State': Sending state 0.00000  with 0 decimals of accuracy
13:11:40	[D]	[text_sensor:067]	
'T-Series Inverter Mode': Sending state 'Offline'
13:11:41	[E]	[api:127]	
No client connected to API. Rebooting...
13:11:41	[I]	[app:127]	
Forcing a reboot...
13:13:31	[D]	[text_sensor:067]	
'T-Series Inverter Mode': Sending state 'Waiting for response...'
13:14:31	[D]	[text_sensor:067]	
'T-Series Inverter Mode': Sending state 'Waiting for response...'
13:15:31	[D]	[text_sensor:067]	
'T-Series Inverter Mode': Sending state 'Waiting for response...'
```

... and it repeats.

I upgraded the code (version with inverter mode parameter) but it also doesn't work. 
I'm new in esphome... what can I check? How to debug it? The wiring is 100% fine because I haven't changed anything.

assembly12, 2022-12-21:
Hi.
Seems like the ESP32 can't  connect to HA for some reason.
Does it reboot every 15 minutes or more often?
Where is you inverter situated. Might it be possible that it got too cold/humid?

The ESP reboots because it encountered some kind of problem (in this case probably the communication to HA).
You could try adding this (for error searching), which stops the ESP from rebooting.

`# Enable Home Assistant API
api:
  reboot_timeout: 0s`

tomaszsluszniak, 2022-12-22:
I am not using Home Assistant to collect the data. I am using Zabbix because of some other things that I have there. 
The inverter is situated in my garage (about 14 degrees Celsius). 
It rebooted more often than 15 minutes.

I put the reboot_timeout parameter as you suggested. And here is the result after flashing the ESP.

```
[08:09:14][C][logger:293]: Logger:
[08:09:14][C][logger:294]:   Level: DEBUG
[08:09:14][C][logger:295]:   Log Baud Rate: 115200
[08:09:14][C][logger:296]:   Hardware UART: UART0
[08:09:14][C][uart.arduino_esp32:108]: UART Bus 1:
[08:09:14][C][uart.arduino_esp32:109]:   TX Pin: GPIO16
[08:09:14][C][uart.arduino_esp32:110]:   RX Pin: GPIO17
[08:09:14][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[08:09:14][C][uart.arduino_esp32:114]:   Baud Rate: 9600 baud
[08:09:14][C][uart.arduino_esp32:115]:   Data Bits: 8
[08:09:14][C][uart.arduino_esp32:116]:   Parity: NONE
[08:09:14][C][uart.arduino_esp32:117]:   Stop bits: 1
[08:09:14][C][template.text_sensor:021]: Template Sensor 'T-Series Inverter Mode'
[08:09:14][C][template.text_sensor:021]:   Icon: 'mdi:solar-power-variant'
[08:09:14][C][captive_portal:088]: Captive Portal:
[08:09:14][C][web_server:125]: Web Server:
[08:09:14][C][web_server:126]:   Address: 192.168.2.6:80
[08:09:14][C][mdns:103]: mDNS:
[08:09:14][C][mdns:104]:   Hostname: foxess-inverter
[08:09:14][C][ota:093]: Over-The-Air Updates:
[08:09:14][C][ota:094]:   Address: 192.168.2.6:3232
[08:09:14][C][ota:097]:   Using Password.
[08:09:14][C][api:138]: API Server:
[08:09:14][C][api:139]:   Address: 192.168.2.6:6053
[08:09:14][C][api:143]:   Using noise encryption: NO
[08:10:01][D][text_sensor:067]: 'T-Series Inverter Mode': Sending state 'Waiting for response...'
[08:11:02][D][text_sensor:067]: 'T-Series Inverter Mode': Sending state 'Waiting for response...'
[08:12:01][D][text_sensor:067]: 'T-Series Inverter Mode': Sending state 'Waiting for response...'
[08:13:01][D][text_sensor:067]: 'T-Series Inverter Mode': Sending state 'Waiting for response...'
[08:14:01][D][text_sensor:067]: 'T-Series Inverter Mode': Sending state 'Waiting for response...'
[08:14:05][I][ota:113]: Boot seems successful, resetting boot loop counter.
[08:14:05][D][esp32.preferences:113]: Saving 1 preferences to flash...
[08:14:05][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[08:14:05][D][sensor:127]: 'T-Series Inverter State': Sending state 0.00000  with 0 decimals of accuracy
[08:14:05][D][text_sensor:067]: 'T-Series Inverter Mode': Sending state 'Offline'
[08:14:06][D][sensor:127]: 'T-Series Inverter State': Sending state 0.00000  with 0 decimals of accuracy
[08:14:06][D][text_sensor:067]: 'T-Series Inverter Mode': Sending state 'Offline'
[08:14:06][D][sensor:127]: 'T-Series Inverter State': Sending state 0.00000  with 0 decimals of accuracy
[08:14:06][D][text_sensor:067]: 'T-Series Inverter Mode': Sending state 'Offline'
[08:14:07][D][sensor:127]: 'T-Series Inverter State': Sending state 0.00000  with 0 decimals of accuracy
...
```

and the last two entries repeat until the ESP restart. The inverter works and the official app shows all the parameters in real-time.

tomaszsluszniak, 2022-12-22:
It's hard to say it, but ... I think that I f..ed up ... I removed this part of yaml definition by mistake:
```yml
switch:
  - platform: gpio
    pin: 4
    name: "T-Series sending mode" #if "on" reading from inverter doesn't work!
    internal: true #switch is not exposed to HA frontend, should you ever need to write to the inverter, just comment this line
```
which I guess is quite important. Now everything is working..

closed 2022-12-22 by tomaszsluszniak