assembly12/Foxess-T-series-ESPHome-Home-Assistant #10: ethernet module

Arturek77, 2023-01-26:
Thanks for the good job. I'm getting ready to implement a project and I have a question. My foxess is connected to the local network via an ethernet module. Will entering my WiFi network details in a yaml file work correctly?

disakos, 2023-01-26:
Hi @Arturek77 you need to follow this tutorial of course if you have wt32-eth01 module.
[https://community.home-assistant.io/t/how-i-installed-esphome-on-the-wt32-eth01/359027](url)
It worked for me.

Arturek77, 2023-01-26:
I think I explained it wrong, I already have the inverter connected with the original ethernet module to the internet. And the ESP32 module I have already bought for WiFi - as in the diagram. And the question is whether it will work because the range is weak. How far can I extend the wiring to the ESP32?

disakos, 2023-01-26:
I think RS-485 is approx. working till around 1km ;) so two cables from Inverter to your ESP can be pretty long. I have Inverter in garage and ESP on 1 floor 10 meters far... working good.

Arturek77, 2023-01-26:
Thanks disakos. So it should work as in the diagram? I just have to enter my wifi details in the yaml file?

closed 2023-04-02 by assembly12