assembly12/Foxess-T-series-ESPHome-Home-Assistant #12: The possibility of usage FoxESS Smart WiFi 3.0 for HA communication

pm90pl, 2023-03-15:
Hi,

Is it possible to use FoxESS Smart WiFi 3.0 dongle for that integration? The dongle does use RS485 communication to get data from the inverter. Wouldn't be possible to connect to the dongle and get the data on demand bypassing FoxCloud?

Cheers,
PM

assembly12, 2023-03-15:
Hi,
do you mean reflashing the FoxESS dongle with ESPHome? It should be possible, as it has an ESP chip inside (I checked ;-)).
I actually thought about doing this, but in the end decided I still wanted to be able to check in to the FoxESS cloud for debugging reasons. If you reflash the dongle, your inverter will no longer be able to communicate with the cloud (or receive firmware updates for that matter).
So in short: I can't see why it wouldn't be possible. It comes with some downsides. And most important; succes isn't guaranteed, try at you own risk ;-)

pm90pl, 2023-03-15:
So without reflashing it won't be possible - would it? What about a ready to use RS485 -> WiFi converters like 
https://www.amazon.co.uk/dp/B07DNWM62H?ref_=cm_sw_r_cp_ud_dp_BPWX7Z53PDES4WJ9JY89
Would they work?

assembly12, 2023-03-18:
If you want to use this (local) integration you would have to flash the dongle. There is however another integeration which reads the data from the Foxess-cloud. In this case you don't have to flash the dongle. The integration doesn't officially support T-Series. I did get it to work, but it was very unstable.
A RS485 to Wifi converter wil work, but keep in mind that you will have to set up all registers in HA manually and it is not compatible with this integration.

closed 2023-04-02 by assembly12