assembly12/Foxess-T-series-ESPHome-Home-Assistant #13: Polling inverter instead of waiting for messages?

pontusfischer, 2023-03-29:
Hi, I have a somewhat related issue. I want to query my T4-G3 inverter using a Modbus extension for my smart home system (called Loxone). If I understand correctly this model of inverter simply sends a status package every 90 seconds whether you want to or not? Can you also ignore this and query the Modbus registers manually? I have the inverter on the same bus as a Fox ESS AIO-H3-8.0 which sort of complicates things :). The AIO can be queried and this is working fine but I need to talk to the T4-G3 as well. I would appreciate any input. Thanks in advance.

assembly12, 2023-04-01:
Hi, as far as I'm aware the T-Series can't be queried. I'm not even sure you can still call the protocol modbus, since there are no registers. The hybrid inverters from Foxess use a different protocol and can be queried, but with T-Series you are stuck the data package which is send every 90 to 120 seconds (depending on the firmware installed on the inverter).
You should still be able to listen to this package, though. The package has the following characteristics:

- starts with 7E7E
- ends with E7E7
- lenght: 165 or 245 bytes (again, depending on firmware)

assembly12, 2023-04-02:
9600

you can take most of it from my code. The .h file is in c++ allready; just ignore the HomeAssistant stuff.

Danielxyz123, 2023-06-10:
Hi, thx for your work.

I tried your code on a G3-T10 and it works, but it seems that the sensorvalues doesn‘t fit all. Is there another byte setting for the 245 byte version? Some values for example in the picture..

[image]

assembly12, 2023-06-10:
Seems like your inverter might have a new firmware, where they added bytes to the header and or footer.
To verify my suspicion you could try to add 4 to the bytes count.

So for example 
`        TwoByte grid_power_value;
        grid_power_value.Byte[0] = bytes[**10**];
        grid_power_value.Byte[1] = bytes[**9**];
        id(grid_power).publish_state(grid_power_value.UInt16);
        delay(50);`

becomes
`        TwoByte grid_power_value;
        grid_power_value.Byte[0] = bytes[**14**];
        grid_power_value.Byte[1] = bytes[**13**];
        id(grid_power).publish_state(grid_power_value.UInt16);
        delay(50);`

You don't have to this to every sensor, just mayby one or two to verify this is actually the issue.
Were you by any means generating around 4200W at the time you took the screenshot?

Danielxyz123, 2023-06-10:
Hi,

thx for your answer.

it doesn‘t seems to be only a shift.
I tried the shift by 4 but it was not correct i think.
 I took some more data to analyse it without shift. Look at the pictures.
[image]

Danielxyz123, 2023-06-10:
[image]

Danielxyz123, 2023-06-10:
[image]

Danielxyz123, 2023-06-10:
[image]

Danielxyz123, 2023-06-10:
[image]
[image]
[image]
[image]

Danielxyz123, 2023-06-10:
[image]
[image]
[image]
[image]
[image]

Danielxyz123, 2023-06-10:
[image]

assembly12, 2023-06-10:
Thanks, those pictures are helpfull! I've been going over the numbers but can't figure it out yet. It definitely isn't far off, though!
If it's not too much trouble, could you try to shift it 3, 5 and 6 as well.
Alternatively you can try to read out the entire message in hex using uart debug.

Danielxyz123, 2023-06-11:
Ok,your are right. It‘s a shift by 6 bytes. It’s a little bit confusion without multiplicator.
Cool, it works 👌🏻. But, why ist „generation power“ zero?

Danielxyz123, 2023-06-11:
[image]

assembly12, 2023-06-11:
hmm, do you have your inverter connected to your meter?
What I'm suspecting now, is that they didn't add anything to the footer, but instead added some sensors between "loads power" and "grid_voltage_r". Which would mean for every sensor before  "grid_voltage_r" you would have to remove the shifting, or apply a different shifting (however always less than 6).
See if you can figure it out, it's hard for me to do, since I don't know what "generation_power" should be at the time of the screenshot. If you're stuck please make a screenshot of the first 4 sensors and the expected value for "generation_power" and I'll see if i can figure it out.

Danielxyz123, 2023-06-11:
Yes the meter is connected...
I think you are right. Unknown 1-3 are the shifted bytes.

succesfully read realtime data
User data length: 238
Total message length: 251
unknown1
-1
unknown2
-6365
unknown3
0
grid_power
6678
generation_power
0
loads_power
313
grid_voltage_r
2335
grid_current_r
95
grid_frequency_r
4998
grid_power_r
2216
grid_voltage_s
2333
grid_current_s
96
grid_frequency_s
4998
grid_power_s
2237
grid_voltage_T
2307
grid_current_T
96
grid_frequency_T
4998
grid_power_T
2212
pv1_voltage
5594
pv1_current
113
pv2_voltage
2876
pv2_current
11
pv3_voltage
0
pv3_current
0
pv4_voltage
0
pv4_current
0

Danielxyz123, 2023-06-11:
[image]

Danielxyz123, 2023-06-11:
[image]

assembly12, 2023-06-11:
ok, I understand now what they have changed.
Try these bytes values for the first three sensors and it should give you the desired results.

    TwoByte grid_power_value;
    grid_power_value.Byte[0] = bytes[13];
    grid_power_value.Byte[1] = bytes[11];
    id(grid_power).publish_state(grid_power_value.UInt16);
    delay(50);

    TwoByte generation_power_value;
    generation_power_value.Byte[0] = bytes[16];
    generation_power_value.Byte[1] = bytes[15];
    id(generation_power).publish_state(generation_power_value.UInt16);
    delay(50);

    TwoByte loads_power_value;
    loads_power_value.Byte[0] = bytes[20];
    loads_power_value.Byte[1] = bytes[19];
    id(loads_power).publish_state(loads_power_value.UInt16);
    delay(50);