assembly12/Foxess-T-series-ESPHome-Home-Assistant #14: Regarding compatibility with F and S series

pablonajarrofever, 2023-04-24:
Hi,

In the project's wiki you mention that, though not tested, it probably works also in F and S-series.

However, the pinout for the invertor connector seems to be different, as can be read in the User Manuals of the different series (https://www.fox-ess.com/downloads/#1608629474343-a4d4beb2-71e3; page 16 of F Series User Manual, for instance).

In F and S-series pin 7 is “NC” (not connected), it doesn’t supply the +12V as in T.  And pins 1 and 2 are used for CT (the Current Transformer sensor of the AC line); so we cannot access RS485 through these pins.
Pins 3 and 4 are, in all three series, assigned to Meter 485. I wonder if their usage is restricted to the connection of a limiting meter, as explained in the manuals, or they are also available for plain RS485 connection. That would allow me to use this setup, though wiring (and powering of circuits) should be different.

Thanks and regards

assembly12, 2023-04-25:
Hi,
yes sadly you are correct. It probably doesn't work with F and S series. At the time I thought the pins labeled with 485- and 485+ would broadcast the serial connection of the inverter. But as it seems, these contacts are meant to be connected to an export limiting meter.
I will remove the suggestion that this might work with F and S series from the wiki.
Thanks for bringing this to my attention and I'm really sorry if I caused you to have false hope!
You might still be able to tap into the RS485 line of the wifi dongle, but thats something I'm not willing to try out for myself at this moment.

Eng65745, 2023-04-28:
Hi,

I also have a Photovoltaic installation with a  FoxESS F-Series inverter installed recently, and have just found this thread. Could you clarify the point of "tapping into the RS485 line of the wifi dongle" in the previous post? Does the dongle connect to the inverter through a RS485 connection? or does it provide a RS485 line itself  ? (how?, where?)

Let me cling to the hope you so kindly mentioned above...

Regards

assembly12, 2023-05-01:
Hi, the dongle probably connects to the inverter via RS485. I don't have an F-Series, so I can not say for sure. Even if you are able to connect to this serial, the protocol on this connection must be same. So no guarantee of succes.
One way to go about it, might be the following. If you can get hold of a second wifi dongle (for cheap), you could flash it with esphome. If you're lucky it should work right away. But still no guarantees whatsoever. Only downside is, you can't connect to the Foxess cloud anymore.

SibrenVasse, 2023-05-01:
Hi!
I also have a F-series inverter. I have captured the network traffic between the WIFI dongle and the cloud server.
One of the data packets that I captured seemed to match the specification in the header file. (Size 165 and 0x7E7E ... E7E7)
Next I tried decoding some fields that were of interest to me: pv1_voltage_value, pv1_current_value, today_yield_value, total_yield_value. The resulting values all matched those displayed in the app.
So the protocol seems to be at least very similar. 

The image in the manual suggests that the dongle is connected by 4 pins to the inverter. So probably those would be two data lines, VCC (5v/12v?) and ground? This would suggest it would be possible to wire the hardware setup on the wiki to this port.

SibrenVasse, 2023-05-01:
Update:
The dongle has 4 connectors. On the connector they are labeled 1-4.
Inside, on the PCB they are labeled:
1: +
2: - 
3: A
4: B

There is a SMD ESP-WROOM-02U inside.

Eng65745, 2023-05-02:
Hi,

This is to confirm the assumption by assembly12 that wifi dongle connects to inverter by RS485, as stated in a FoxESS official datasheet, product specifications, last line: "Communications Method: RS485". 

[smart-wifi datasheet.pdf](https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant/files/11371256/smart-wifi.datasheet.pdf)

Thanks to both assembly12 and SibrenVasse, your posts are very helpful to achieve data collection directly from the inverter instead of FoxESS Cloud that, by the way, seems to experience frequent outages not being accessible for several hours (though it keeps collecting data). Last time, yesterday May 1st.

assembly12, 2023-05-02:
ok. That looks very promising!
It looks like this repo will work on the wifi (usb) bus as well. The hardware setup as described in the wiki should work, only the connector has to be swapped to usb.
If anyone is willing to try this out, im happy to assist.

SibrenVasse, 2023-05-02:
Just to be clear: the wifi dongle is not connected via the USB port of the device! The USB port is only for mass storage devices for a fimware upgrade of the inverter.

The connector is screwed into the inverter with a connector that looks like this one:
https://m.media-amazon.com/images/I/61-1h5O01KL._AC_UF1000,1000_QL80_FMwebp_.jpg
The dongle is the female side, the inverter has the male connector.

I'm looking into the exact size of this connector and see if I can order one and try it with the hardware described on the wiki.

Eng65745, 2023-05-02:
In case it helps: I found these drawings with the exact sizes of the connectors SibrenVasse show (it seems they are commonly used for microphones and in aviation, in a local store they cost between 2 and 3 euros). I attach two versions of the male connector, the one to mount on a chassis and the one to attach to a cable (exactly the same dimensions as the female, it seems).

[image]
[image]
[image]

Eng65745, 2023-05-02:
Sorry, dimensions for male and female cable connectors are NOT the same (they shouldn't or they won't fit!)

Eng65745, 2023-06-17:
Hello,

I had the chance to measure the Wifi dongle and compare to the audio-type connectors shown by SibrenVasse or myself in previous posts.

The body of the dongle connector is sligthly larger (by 0.2 mm, aprox) than the audio-type connectors. In fact, I tried to plug one audio-type connector into the inverter and it fits, pins match, but the slight difference leaves the connector a bit loose; in vertical position it would slip out. Maybe a couple of turns of tape would make for the difference and the connector would stick.

The real difference is on the fastening nut; the dongle's nut is 22.4 mm in diameter, the audio-type connector's nut 18 mm. 

Also, labelling of pins is different - pairs 1 & 2 and 3&4 are reversed.

In summary: one of these conventional audio-type connectors can be plugged into the inverter, pins will match, but it cannot be fastened, so it would require some element to keep it firmly in place.

SibrenVasse, 2023-06-17:
I have ordered this connector: https://nl.aliexpress.com/item/1005005256853917.html
This connector fits on the inverter and you can tighten the screw. The only thing is the indentation on the side of the connector is a little bit too small, so disconnecting it from the inverter is a little bit difficult. You can probably widen this indentation by filing it down to make it fit better.

The listing on AliExpress has become much more expensive since I ordered.

Eng65745, 2023-06-23:
Hello SibrenVasse,

I've noticed the reference from aliexpress is a package of 10 pairs of male-female connectors. If after you've completed your project you don't have an use for some of them, I'd be glad to purchase a pair from you and help you to recover costs.

SibrenVasse, 2023-06-23:
I'm sorry but there used to be an option to only buy one.
So I only ordered one connector!

SibrenVasse, 2023-07-11:
Seems like the lines that output 12V when the dongle is connected, do not supply VCC when I only connect the buck converter. Any ideas?

SibrenVasse, 2023-07-16:
Success! My ESP is connected to the inverter and is supplying Home Assistant with data!

A few notes:
- The Inverter will only supply Pin 2 with VCC when the RS485 module is detected.
- I have seen several voltages between Pins 1 and 2 ranging from 7V to 12V. Because of this, for the moment I use a USB cable to provide the board with 5V. So I only have Pins 2, 3 and 4 connected to the inverter (GND to provide common ground, A and B), and the USB cable is connected to Pins 1 and 2 (VCC and GND) to provide power. 

[image]

assembly12, 2023-07-18:
Very well done!
So to verify; you are using the exact same code, or have you altered anything?

So it should probaly be possible to flash the orignal Wifi dongle with ESPhome as well. Perhaps that's an option for the hardware issues (expensive connector, power supply).

SibrenVasse, 2023-07-18:
Yes, exactly the same code. No alterations. I only changed the  SSID etc. in the .yml file.

Flashing the original dongle is definitely something that should be possible.  It looks like there is a TTL 'port' on the PCB. (Four unpopulated holes)

bubupol, 2023-07-23:
Great Work ! Thank you for Sharing.

Have you tried The Wifi dongle and the ESP32 connected at the same time ? 
(that keeps the FOXESS cloud data as back up and software upgrade possible, alarms etc...)

SibrenVasse, 2023-07-23:
I have not. I think RS485 should be able to handle multiple devices.

Eng65745, 2023-07-24:
I want to add my thanks to SibrenVasse for his contribution to this topic. Now it's possible to feed inverter data directly to our HA (or other platforms) installation.

Also it opens the door to other possibilities: hacking the wifi dongle to send data directly to HA instead of to FoxESS cloud, as assembly12 suggested (so no extra hardware would be neccesary). Or connecting both  wifi dongle and our local ESP-based hardware (it would requires some expert knowledge on how RS-485 multidrop works), so we feed both our HA and FoxESS cloud, as bubupol mentions. Or using our local ESP to send data both to our HA and FoxESS cloud (keeping the manufacturer connected could be important for support and warranty issues).

Thanks a lot to all who show the way to those of us with less skills!

bubupol, 2023-07-25:
Hello,
success too ! on a F-series inverter (F6000). The data is received every 90 seconds.
For now i used an extra power supply for testing, 
I wired it in parallel to the wifi dongle. so both are working.

thank you all.

velk101, 2023-07-27:
Also working on a F3600 with the same delay of about 90 seconds, just connect to the solder pads black and white wires rs485

[image]
[image]

mattwilson1066, 2023-08-09:
> 

When you say you wired in parallel, have you got any photos of how you did this? do you mean you have the original dongle connected, but then taken A/B feed from the dongle? And do you think that would work by then connecting directly into something like a https://www.waveshare.com/usb-to-rs485-b.htm
I currently use one to connect to my H1, so could try that or get another and wire up the F4600 into a 2nd one.

velk101, 2023-08-10:
If you open up the dongle with the two screws, you'll see the four wires soldered to the PC
Connect to the white wire which is A on the RS485
Connect to the black wire which is B on the RS485
You then can run the WiFi and have an RS485 to your PC

velk101, 2023-08-11:
I drilled a small hole in the bottom of the case for four wires to come out, i then fitted a small 4way connector so i could disconnect it  when i got bored with it lol

mattwilson1066, 2023-10-17:
Hi All, so I have drilled the hole and have now connected the A/B cables to  https://www.waveshare.com/usb-to-rs485-b.htm
Is anyone able to advise which integration I should use to access the data via this method? I can see there is mention of the ESP board, but does the code need modifying to work with USB to RS485 board? Sorry this is a little out of my ability, but am happy to try and report back. Thanks All

assembly12, 2023-12-17:
Hi, this protocol doesn't allow polling. You are bound to the 1.5-2 minute update inverval.

jepminer, 2026-09-07:
Just having had 2 S G2 inverters installed this is of great interest to me, the best way to go about it seems to be the final method where you intercept the signal so you can keep the WiFi connection and read the data via the esp32 however the other 2 cables on the dongle I assume those are power, is there 5 or 12v there that the esp32 can be powered off? I have WW loggers as well if that is the one being used here.