assembly12/Foxess-T-series-ESPHome-Home-Assistant #15: Latest firmware broke integration

giedriuskilcauskas, 2023-08-03:
It seems that latest firmware broke integration. most of the fields seems to be showing weird values:
[image]
These are versions currently installed:
[image]

@assembly12 ,Would you mind looking into that?

Thanks,
Giedrius

hartmut58, 2023-08-06:
Hi,
it is a firmware problem. Downgrade to manager 1.20 and it works.
Hartmut

Tanil2008, 2023-08-06:
Hi. I have the same problem even though the manager's version is 1.20. Please help.
[image]

assembly12, 2023-08-06:
Hi,
i'm working on adding support for the new firmware (new protocol).
[](https://community.home-assistant.io/t/foxess-inverter-modbus/459313/164?u=assembly)
I'm allmost there, just waiting for the confirmation on the generation power.

giedriuskilcauskas, 2023-08-06:
> Hi, i'm working on adding support for the new firmware (new protocol). [Link to HA Forum](https://community.home-assistant.io/t/foxess-inverter-modbus/459313/164?u=assembly) I'm allmost there, just waiting for the confirmation on the generation power.

I confirm that 16/15 shows generation power. Without shifting that was grid power previously, not sure where it is now

hartmut58, 2023-08-07:
Hello,
here is a Node-Red export, the addresses are not fully checked. But for my purposes (iobroker) it is enough.
Hope you can do something with it.

Greetings hartmut

{
                "type": "int16be",
                "name": "GridPower",
                "offset": 15,
                "length": 1,
                "offsetbit": 0,
                "scale": "1",
                "mask": ""
            },
            {
                "type": "int16be",
                "name": "GenerationPower",
                "offset": 17,
                "length": 1,
                "offsetbit": 0,
                "scale": "1",
                "mask": ""
            },
            {
                "type": "int16be",
                "name": "LoadsPower",
                "offset": 19,
                "length": 1,
                "offsetbit": 0,
                "scale": "1",
                "mask": ""
            },
            },
            {
                "type": "int16be",
                "name": "GridCurrent_R",
                "offset": 23,
                "length": 1,
                "offsetbit": 0,
                "scale": "0.1",
                "mask": ""
            },
            {
                "type": "uint16be",
                "name": "GridFrequency_R",
                "offset": 25,
                "length": 1,
                "offsetbit": 0,
                "scale": "0.01",
                "mask": ""
            },
            {
                "type": "int16be",
                "name": "GridPower_R",
                "offset": 27,
                "length": 1,
                "offsetbit": 0,
                "scale": "1",
                "mask": ""
            },
            {
                "type": "uint16be",
                "name": "GridVoltage_S",
                "offset": 29,
                "length": 1,
                "offsetbit": 0,
                "scale": "0.1",
                "mask": ""
            },
[... 398 lines cut ...]
                "name": "FaultMessage_DWORD8",
                "offset": 159,
                "length": 1,
                "offsetbit": 0,
                "scale": "1",
                "mask": ""
            },
            {
                "type": "uint16be",
                "name": "Master State",
                "offset": 163,
                "length": 1,
                "offsetbit": 0,
                "scale": "1",
                "mask": ""
            },
            {
                "type": "uint16be",
                "name": "PV Input Number",
                "offset": 165,
                "length": 1,
                "offsetbit": 0,
                "scale": "1",
                "mask": ""
            }

hartmut58, 2023-08-07:
Hello,
just tested -
15/16 generation power
175/176 grid power

Greetings Hartmut

Tanil2008, 2023-08-08:
Here is the effect of my tests:
[foxess_t_series.zip](https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant/files/12294568/foxess_t_series.zip)
To determine the bits T-Series Grid Current R, T-Series Grid Current S, T-Series Grid Current T - because they show bad data.
My inverter: Foxess T5 G3, manager 1.20.

hartmut58, 2023-08-10:
[cid:7c305820-fbfb-4b01-b92a-da9eb3c649e4]

Hello,
I think so it is ok, please test.

#include "esphome.h"

unsigned long millis_lastmessage = 0;
const long inverter_timeout = 300000;
int inverter_mode = 99; //0=Offline, 1=Online, 2=Error, 99=Waiting for response...

class foxesscomponent : public PollingComponent, public Sensor, public UARTDevice {
  public:
    foxesscomponent(UARTComponent *parent) : PollingComponent(600), UARTDevice(parent) {}
  
  Sensor *frame_header = new Sensor;
  Sensor *function_code = new Sensor;
  Sensor *time_stamp = new Sensor;
  Sensor *data_length = new Sensor;
  Sensor *grid_power = new Sensor;
  Sensor *generation_power = new Sensor;
  Sensor *loads_power = new Sensor;
  Sensor *grid_voltage_r = new Sensor;
  Sensor *grid_current_r = new Sensor;
  Sensor *grid_frequency_r = new Sensor;
  Sensor *grid_power_r = new Sensor;
  Sensor *grid_voltage_s = new Sensor;
  Sensor *grid_current_s = new Sensor;
  Sensor *grid_frequency_s = new Sensor;
  Sensor *grid_power_s = new Sensor;
  Sensor *grid_voltage_T = new Sensor;
  Sensor *grid_current_T = new Sensor;
  Sensor *grid_frequency_T = new Sensor;
  Sensor *grid_power_T = new Sensor;
  Sensor *pv1_voltage = new Sensor;
  Sensor *pv1_current = new Sensor;
  Sensor *pv1_power = new Sensor;
  Sensor *pv2_voltage = new Sensor;
  Sensor *pv2_current = new Sensor;
  Sensor *pv2_power = new Sensor;
  Sensor *pv3_voltage = new Sensor;
  Sensor *pv3_current = new Sensor;
  Sensor *pv3_power = new Sensor;
  Sensor *pv4_voltage = new Sensor;
  Sensor *pv4_current = new Sensor;
  Sensor *pv4_power = new Sensor;
  Sensor *boost_temperature = new Sensor;
  Sensor *inverter_temperature = new Sensor;
  Sensor *ambient_temperature = new Sensor;
  Sensor *today_yield = new Sensor;
  Sensor *generation_total = new Sensor;
  Sensor *from_grid_yield_generation = new Sensor;
  Sensor *inverter_state = new Sensor;

  void setup() override {
    id(inverter_state).publish_state(inverter_mode);
    millis_lastmessage = millis();
  }

  std::vector<int> bytes;

  //void loop() override {
  void update() {
    if(millis_lastmessage + inverter_timeout < millis()) {
      if(inverter_mode != 0){
        inverter_mode = 0; //offline
        id(inverter_state).publish_state(inverter_mode);
        id(generation_power).publish_state(0);
        id(grid_current_r).publish_state(0);
        id(grid_power_r).publish_state(0);
[... 299 lines cut ...]
        delay(50);
        uint32_t pv3_power_value = pv3_current_value.UInt16 * pv3_voltage_value.UInt16 * 0.01;
        id(pv3_power).publish_state(pv3_power_value);
        delay(50);
        uint32_t pv4_power_value = pv4_current_value.UInt16 * pv4_voltage_value.UInt16 * 0.01;
        id(pv4_power).publish_state(pv4_power_value);
        delay(50);

        id(inverter_state).publish_state(inverter_mode);

        bytes.clear();
      }
      else {
      }
    }
  }

  typedef union
  {
    unsigned char Byte[2];
    int16_t Int16;
    uint16_t UInt16;
    unsigned char UChar;
    char Char;
  }TwoByte;};

Danielxyz123, 2023-08-12:
Did someone test the latest Manager Version 1.22?

hartmut58, 2023-08-12:
With Manager Firmware 1.22 the RS485 port is dead, just like with version 1.21.

Danielxyz123, 2023-08-12:
ok, thanks for that information. Is a downgrade possible?

hartmut58, 2023-08-13:
Downgrade is possible via USB, rename ...1.20 to ...1.22 (no guarantee, can of course go wrong)

Danielxyz123, 2023-08-13:
Ok, where do i get the File? Only from Support?

hartmut58, 2023-08-13:
Yes, open a ticket with support, otherwise Foxess will not notice the bug.

mekket, 2023-08-18:
> Does somebody not this project?
> 
> https://github.com/peper0/foxess-t-modbus-ha
> 
> Seems that Modbus is recently possible ?

This approach promises almost real time data and if I understood correctly it does it because it is connected to the meter RS485 side not the data RS485 that assembly12-s version uses. My T10 G3 shows almost real time data on its display so the inverter must have it available somehow. I just got my system running today and I installed a smart meter but I have not been able to get my ESP and home assistant working as intended. Possibly because of the firmware issue? I am on manager 1.21. Anyways I had a thought about connecting the ESP to the Meter connections alongside the meter itself. Would it work or does the RS485 not work like that? If someone here has a smart meter and has this integration working - maybe you could try it out and let us know. I am already sick of seeing the 5 minute old data from FoxEss Cloud :(

mekket, 2023-08-18:
Do You also have a T10 G3? Is assemblys version or some modified version of it working for you? Did you downgrade your firmware?

Danielxyz123, 2023-08-18:
Yes, i have s T10 G3. Which Firmware do you use? It worked until i used manager 1.22 🙈. I‘m working on an downgrade. But currently no success. The support sends me some Files but it dont‘t work yet..

Danielxyz123, 2023-08-22:
> So I understand right that manager 1.22 kills access to RS485 and it's no longer working. No data coming in at all or it's somewhat garbage? I currently have 1.21 and with above changes to addressing, all works fine.

Yes, it seems that 1.22 does deactivate that “sending data” on rs485…

I don’t know if it’s a bug. Downgrade didn’t work for me yet…

hartmut58, 2023-08-22:
For me too, it seems that Manager 1.21 and Manager 1.22 disables the sending of data. (Inverter T10-G3)
The German support sent me the manager 1.20 for a downgrade. For a successful downgrade, I had to rename manager1.20 to manager1.22, otherwise the inverter rejects the upgrade with the message "wrong file".
The support in China does not give any answer.
Unfortunately there is no changelog from FOXESS to the changes of the firmware.

Danielxyz123, 2023-08-22:
Thx for that Information. How did you change the Filename exactly? My Filename is attached. Only the ….V1.20 to ….V1.22?

[image]

hartmut58, 2023-08-22:
Original / 10-300-11040-xx_T30_Manager_V1.20.bin
New /       T30_Manager_V1.22.bin

Danielxyz123, 2023-08-22:
Thx, a lot. It works 👌🏻

assembly12, 2023-08-22:
Hi guys! Sorry i've been quiet for a while, I had some personal business to attend to.

Addressing some of the topics that were discussed in this issue:

1.  I didn't know about the other integration which presumably connects to a "real" modbus. As far as I could figure out, it doesn't work with my inverter (T12 G3). Updates every few seconds would be great, but G3 doesn't seem to be compatible with this setup.

2. Regarding the compatibiltiy issues with newer firmwares: this is exactly the reason why I do not upgrade the firmware :-). I'm very happy, both with the way the inverter is running and it's integration in HA as well. I see, as of now, no need to upgrade and have in fact now removed the foxess dongle entirely. That being said I cannot test anything for the newer versions.

3. Regarding 1.22: as far as I understand the RS485 communication is entirely terminated in this version. Thanks to hartmut58 for giving us a working solution for downgrading to 1.20 again!!

4. Regarding 1.21: with the help of "xndz" over at the HA forums we seem to have the new registernumbers for 1.21 decoded. I'll try implement these registers in the code, so it works with both 1.20 and 1.21, or just release a new version for 1.21 alltogether. I'm not sure when I will be able to do this, though.

Danielxyz123, 2023-08-23:
> Should the RS485 converter flash leds or show some sign of communication? Im on Manager 1.21, Master 1.27 and Slave 1.01. T10 G3. My inverter is in my basement so the wifi signal there is not the best but I have tried twice and it seems that both times when I connect my ESP to the inverter it will not even connect to my wifi and I think it brings down the cloud connection as well because the inverter status goes to Offline shortly after I connect the ESP. It will come back to normal when I remove it. The ESP was able to communicate "blank" values to home assistant when it was on my table. I am using the registry numbers from xndz who has it working on 1.21.
>
> I don't even know where to start my troubleshooting because I can not be sure that my inverter is even able to communicate over RS485. The RED power indicator LED is on but there seems to be another one next to it and that one is constantly off - should it flash when working normally?

It doesn‘t work with 1.21.. 
Last working is 1.20… thats what hartmut also said.

giedriuskilcauskas, 2023-08-24:
> It doesn‘t work with 1.21.. Last working is 1.20… thats what hartmut also said. Von meinem iPhone gesendet
> […](#)

Well, 1.21 clearly works fine for me

giedriuskilcauskas, 2023-08-24:
> Should the RS485 converter flash leds or show some sign of communication? Im on Manager 1.21, Master 1.27 and Slave 1.01. T10 G3. My inverter is in my basement so the wifi signal there is not the [...]

Inverter going offline would indicate something like short circuiting wrong pins or so. For me the basic rs485 adapter didn't work, so I've got a bit more advanced one with automatic flow control and I also power ESP via usb (so don't use step-down from 12V)

mekket, 2023-08-24:
The inverter itself stays online and keeps working, the communication seems to go down even for the cloud. I will try again tomorrow and if that is still the case it must be either my connections or the fact that we have 3 different people with 1.21 with 3 different results.

mekket, 2023-08-29:
I re-flashed my ESP and now it is working as intended(On manager 1.21) Is there a way to change the inverter settings with this approach? Like changing the export limit with HA automations?

Danielxyz123, 2023-08-29:
Does somebody know, which function has manager, master, slave software for T-Series? Is standalone only manager relevant?

kuminalli, 2024-02-13:
> For me too, it seems that Manager 1.21 and Manager 1.22 disables the sending of data. (Inverter T10-G3) The German support sent me the manager 1.20 for a downgrade. For a successful downgrade, I had [...]

Could you please share V1.20 bin-file?

Danielxyz123, 2024-02-14:
[10-300-11040-xx_T30_Manager_V1.20.zip](https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant/files/14285202/10-300-11040-xx_T30_Manager_V1.20.zip)

Here you are, unzip...  .bin files are not supported.

mekket, 2024-02-14:
Has anyone tried setting the export limit from home assistant using rs485? In my area the electricity prices sometimes go to negative values and exporting to grid would actually cost money to the exporter so it would be neat to keep the inverters export limit near the current house consumption which I have from the smart meter.

kuminalli, 2024-02-15:
> <assembly12/Foxess-T-series-ESPHome-Home-Assistant/issues/15/1944456953@
> github.com>

Thank You Very Much! 🙂

hJcRox, 2024-07-30:
Hi
For info, I upgraded to 1.24 before reading this thread and the RS485 stopped working.
Thanks to the good info here I managed to downgrade to 1.20 and RS-485 now works again.
I had to rename the firmware file to 1.24 according to described above.
Interestingly, in the firmware view in FoxEssCloud, it still states revision 1.24 for the manager but 1.20 is running.
It seems like they are using the file name for the displayed firmware version.
Regards Jan

Lembi203, 2024-12-09:
Where do you find the Manager Version?

My inverter is new and has also a different connector with more pins, but i cannot find the manaer version?
 
[image]
[image]

hJcRox, 2024-12-13:
Hi
The 1.20 FW is provided in this thread by Danielxyz123.
[Link](https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant/issues/15#issuecomment-1944456953)
Regards