assembly12/Foxess-T-series-ESPHome-Home-Assistant #16: 3D Enclosure

sweti3, 2023-08-15:
Hi guys,

did someone already print out a case fitting for this board? I am struggling with this..

thx!

Sweti934, 2023-08-22:
With a design for the housing I think I am just stupid- and thought maybe someone has already figured something out 👌

assembly12, 2023-08-22:
Ah, ok i see ;-)
Are you using my design of the pcb? I might be able to design a case for this at some point.

Sweti934, 2023-08-22:
Thanks so much! Yeah it is exactly your pcb with components I got from the Amazon here in Germany 🇩🇪

thesame-, 2023-10-31:
Hi! Thank you for the work you have put into this project @assembly12. Have you had any time for the case design?

assembly12, 2023-11-01:
Thanks! I haven't gotten around to it yet, but it's still on my todo list ;-)

spitfire, 2024-10-18:
> Hi! Thank you for the work you have put into this project @assembly12. Have you had any time for the case design?

Im using Lily T-CAN485 board - which has a RS485, CAN port and power supply accepting 5-12V - so it can be used with the inverter directly without any soldering, it also has an RGB LED if you want it to show status of the inverter. There are multiple cases available for it to print