assembly12/Foxess-T-series-ESPHome-Home-Assistant #18: PCB for that integration

kamyk, 2023-12-21:
Hello,

Since yesterday many people have a problems with an access to Foxcloud by HA. I have all parts and I would like to do it asap. @assembly12  do you maybe have that PCB to sell somewhere?

Thank you!
Kamyk

assembly12, 2023-12-21:
Hi,
yes I have some spares lying around. If you live in Germany I can send you one. Otherwise it will probably be cheaper to order them staight from China ;-)
You can send my a pm over at the [HA Forum](https://community.home-assistant.io/u/assembly) if you like.

kamyk, 2024-01-03:
> Hi, yes I have some spares lying around. If you live in Germany I can send you one. Otherwise it will probably be cheaper to order them staight from China ;-) You can send my a pm over at the [HA Forum](https://community.home-assistant.io/u/assembly) if you like.

Hi :) I live in Berlin :) We can do it! Let me know what should I send to you and how to communicate by mail.

Best,
Rafal

kamyk, 2024-01-03:
> Hi, yes I have some spares lying around. If you live in Germany I can send you one. Otherwise it will probably be cheaper to order them staight from China ;-) You can send my a pm over at the [HA Forum](https://community.home-assistant.io/u/assembly) if you like.

Sent PM already. Thank you!

spielkind1611, 2024-08-02:
@assembly12 
Do you perhaps still have a PCB lying around that you no longer need?

assembly12, 2024-08-03:
Yeah, I've got some left. Since Private messaging isn't possible on github, please contact me on the Home Assistant forum.

spielkind1611, 2024-08-04:
How can i contact you?

SibrenVasse, 2024-08-04:
My email is listed on my github profile.