assembly12/Foxess-T-series-ESPHome-Home-Assistant #21: Improvement idea: structure repo to work as an external_components for ESPHome

spitfire, 2023-12-28:
https://esphome.io/components/external_components.html#example-of-local-components

SibrenVasse, 2024-07-23:
I've started working on this, you can check my progress on my fork: https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant

spitfire, 2024-07-24:
> I've started working on this, you can check my progress on my fork: https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant

Hey, first of all thanks for working on that!
I've tried to migrate over by changing 

```
includes:
     - templates/Foxess-T-series-ESPHome-Home-Assistant/foxess_t_series.h
```

from this repo to

```
external_components:
  - source:
      type: git
      url: https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant    
```

shown in your example yaml.

Unfortunately it spawned this error at compile time:

```
/config/foxess-inverter.yaml: In lambda function:
/config/foxess-inverter.yaml:122:32: error: expected type-specifier before 'foxesscomponent'
     auto foxesssensors = new foxesscomponent(id(uart_bus));
                                ^~~~~~~~~~~~~~~
/config/foxess-inverter.yaml:158:7: error: could not convert '{<expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>, <expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::sensor::Sensor*>'
     };
       ^
```

which seem to be related to lambda function from https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant/blob/main/foxess-inverter.yaml (line 61-100 here).

Is there any way to cleanly replace that? I've seen your https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant/blob/main/foxess-inverter.yaml.example, but I'm afraid it would completely replace my entity names and anything depending on them in the process.

SibrenVasse, 2024-07-24:
If you're worried about entities changing name in Home Asisstant, you have to make sure the new sensor has the same 'name' field in the yaml file. Like this:

```
button:
  - platform: restart
    name: "Foxess inverter restart"

text_sensor:
  - platform: template
    name: "T-Series Inverter Mode"
    id: "inverter_mode_text_sensor"
    icon: mdi:solar-power-variant

sensor:
  - platform: foxess_solar
    phase_a:
      voltage:
        name: "T-Series Grid Voltage R"
      current:
        name: "T-Series Grid Current R"
      active_power:
        name: "T-Series Grid Power R"
      frequency:
        name: "T-Series Grid Frequency R"
    phase_b:
      voltage:
        name: "T-Series Grid Voltage S"
      current:
        name: "T-Series Grid Current S"
      active_power:
        name: "T-Series Grid Power S"
      frequency:
        name: "T-Series Grid Frequency S"
    phase_c:
      voltage:
        name: "T-Series Grid Voltage T"
      current:
        name: "T-Series Grid Current T"
      active_power:
        name: "T-Series Grid Power T"
      frequency:
        name: "T-Series Grid Frequency T"

    pv1:
      voltage:
        name: "T-Series PV1 Voltage"
      current:
        name: "T-Series PV1 Current"
      active_power:
        name: "T-Series PV1 Power"
    pv2:
      voltage:
        name: "T-Series PV2 Voltage"
      current:
        name: "T-Series PV2 Current"
      active_power:
        name: "T-Series PV2 Power"
    pv3:
      voltage:
        name: "T-Series PV3 Voltage"
      current:
        name: "T-Series PV3 Current"
      active_power:
        name: "T-Series PV3 Power"
    pv4:
      voltage:
        name: "T-Series PV4 Voltage"
      current:
        name: "T-Series PV4 Current"
      active_power:
        name: "T-Series PV4 Power"

    total_energy_production:
      name: "T-Series Generation Total"
    energy_production_day:
      name: "T-Series Today Yield"

    generation_power:
      name: "T-Series Generation Power"
    grid_power:
      name: "T-Series Grid Power"
    loads_power:
      name: "T-Series loads Power"

    inverter_temp:
      name: "T-Series Inverter Temperature"
    ambient_temp:
      name: "T-Series Ambient Temperature"
    boost_temp:
      name: "T-Series Boost Temperature"

    inverter_status:
      name: "T-Series Inverter State"
      internal: true
      on_value:
        then:
          - lambda: |-
              if (x==0) {
                id(inverter_mode_text_sensor).publish_state("Offline");
              } else if (x==1) {
                id(inverter_mode_text_sensor).publish_state("Online");
              } else if (x==2) {
                id(inverter_mode_text_sensor).publish_state("Error");
              } else {
                id(inverter_mode_text_sensor).publish_state("Waiting for response...");
              }

# Uncomment this sensor and the i2c section to use the AHT10
#  - platform: aht10
#    temperature:
#      name: "Foxess inverter ambient Temperature"
#    humidity:
#      name: "Foxess inverter ambient Humidity"

#i2c:

```

spitfire, 2024-10-18:
> I've started working on this, you can check my progress on my fork: https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant

@SibrenVasse for some reason your version fails to update some of the sensors, while the version from this repo does. I've noticed my Foxess Inverter T-Series Today Yield (and Total) was updating very infrequently (less often than once an hour) messing up my statistics.
```
[12:31:10][W][component:170]: Component foxess_solar.sensor cleared Warning flag
[12:31:10][W][FoxessSolar::parse_message:134]: Unexpected msg length, length: 165
[12:31:10][W][component:157]: Component foxess_solar.sensor set Warning flag: unspecified
[12:31:10][D][sensor:094]: 'T-Series Grid Power': Sending state 0.00000 W with 0 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Generation Power': Sending state 3576.00000 W with 0 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series loads Power': Sending state 0.00000 W with 0 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Voltage R': Sending state 249.50000 V with 1 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Current R': Sending state 4.70000 A with 1 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Frequency R': Sending state 49.98000 Hz with 2 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Power R': Sending state 1192.00000 W with 0 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Voltage S': Sending state 252.50000 V with 1 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Current S': Sending state 4.80000 A with 1 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Frequency S': Sending state 49.98000 Hz with 2 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Power S': Sending state 1192.00000 W with 0 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Voltage T': Sending state 249.20000 V with 1 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Current T': Sending state 4.80000 A with 1 decimals of accuracy
[12:31:10][D][sensor:094]: 'T-Series Grid Frequency T': Sending state 49.98000 Hz with 2 decimals of accuracy
[12:31:18][D][sensor:094]: 'WiFi Signal dB': Sending state -49.00000 dBm with 0 decimals of accuracy
[12:31:18][D][sensor:094]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[12:31:19][D][esp32.preferences:114]: Saving 1 preferences to flash...
[12:31:19][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[12:31:20][D][internal_temperature:069]: Ignoring invalid temperature (success=0, value=53.3)
[12:32:18][D][sensor:094]: 'WiFi Signal dB': Sending state -47.00000 dBm with 0 decimals of accuracy
[12:32:18][D][sensor:094]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[12:32:20][D][internal_temperature:069]: Ignoring invalid temperature (success=0, value=53.3)
[12:33:10][W][component:170]: Component foxess_solar.sensor cleared Warning flag
[12:33:10][W][FoxessSolar::parse_message:134]: Unexpected msg length, length: 165
[12:33:10][W][component:157]: Component foxess_solar.sensor set Warning flag: unspecified
[12:33:10][D][sensor:094]: 'T-Series Grid Power': Sending state 0.00000 W with 0 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Generation Power': Sending state 3573.00000 W with 0 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series loads Power': Sending state 0.00000 W with 0 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Voltage R': Sending state 253.90001 V with 1 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Current R': Sending state 4.70000 A with 1 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Frequency R': Sending state 49.96000 Hz with 2 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Power R': Sending state 1191.00000 W with 0 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Voltage S': Sending state 251.40001 V with 1 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Current S': Sending state 4.80000 A with 1 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Frequency S': Sending state 49.96000 Hz with 2 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Power S': Sending state 1191.00000 W with 0 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Voltage T': Sending state 251.60001 V with 1 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Current T': Sending state 4.70000 A with 1 decimals of accuracy
[12:33:10][D][sensor:094]: 'T-Series Grid Frequency T': Sending state 49.96000 Hz with 2 decimals of accuracy
[12:33:18][D][sensor:094]: 'WiFi Signal dB': Sending state -48.00000 dBm with 0 decimals of accuracy
[12:33:18][D][sensor:094]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[12:33:19][D][esp32.preferences:114]: Saving 1 preferences to flash...
```

using code from this repo returns all data, all the time (every 2 minutes for my inverter):

```
[12:39:10][I][custom:113]: User data length: 152
[12:39:10][I][custom:114]: Total message length: 165
[12:39:10][D][sensor:094]: 'T-Series Grid Power': Sending state 0.00000 W with 0 decimals of accuracy
[12:39:10][D][sensor:094]: 'T-Series Generation Power': Sending state 3576.00000 W with 0 decimals of accuracy
[12:39:10][D][sensor:094]: 'T-Series loads Power': Sending state 0.00000 W with 0 decimals of accuracy
[... cut ...]
[12:39:12][D][sensor:094]: 'T-Series PV3 Power': Sending state 0.00000 W with 1 decimals of accuracy
[12:39:12][D][sensor:094]: 'T-Series PV4 Power': Sending state 0.00000 W with 1 decimals of accuracy
[12:39:12][D][sensor:094]: 'T-Series Inverter State': Sending state 1.00000  with 0 decimals of accuracy
[12:39:12][D][text_sensor:064]: 'T-Series Inverter Mode': Sending state 'Online'
[12:39:12][W][component:237]: Component <unknown> took a long time for an operation (2119 ms).
[12:39:12][W][component:238]: Components should block for at most 30 ms.
```

SibrenVasse, 2024-10-18:
Can you upload your yaml file? So I can check that everything is the same in my environment.

spitfire, 2024-10-20:
```
substitutions:
  name: "foxess-inverter"
  friendly_name: "Foxess Inverter"
  device_description: "Modbus interface for Foxess Inverter"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: ${device_description}
  # includes:
  #   - templates/Foxess-T-series-ESPHome-Home-Assistant/foxess_t_series.h #Legacy, from https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant

    
esp32:
  board: esp32dev
  framework:
    # type: arduino
    type: esp-idf

    # version: "5.3.1" #src/foxess_t_series.h:351:30: error: format '%i' expects argument of type 'int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
    # platform_version: 6.9.0
  

external_components:
  - source:
      type: git
      url: https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant    

# Enable logging
logger:
  baud_rate: 0
  # level: VERBOSE
  # logs:
  #   light: VERBOSE

<<: !include templates/spitfire/common/api.yaml

<<: !include templates/spitfire/common/ota.yaml

<<: !include templates/spitfire/common/wifi.yaml

<<: !include templates/spitfire/common/webserver.yaml

<<: !include templates/spitfire/common/time.yaml

#bluetooth_proxy:
  #active: true

# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
      allow_other_uses: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
[... 497 lines cut ...]
#     state_class: measurement
#     device_class: power
#     icon: mdi:lightning-bolt
#     accuracy_decimals: 1
    
#   - name: "T-Series Inverter State"
#     id: "inverter_state"
#     internal: true
#     on_value:
#       then:
#         - lambda: |-
#             if (x==0) {
#               id(inverter_mode_text_sensor).publish_state("Offline");
#             }
#             else if(x==1) {
#               id(inverter_mode_text_sensor).publish_state("Online");
#             }
#             else if(x==2) {
#               id(inverter_mode_text_sensor).publish_state("Error");
#             }
#             else {
#               id(inverter_mode_text_sensor).publish_state("Waiting for response...");
#             }
  #Endregion Legacy Foxess code
```

SibrenVasse, 2024-10-20:
Somehow it's only publishing the sensor states until line [foxess_solar.cpp#L156](https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant/blob/ec5977a3af52e2e402e82bb1fbf97edea416ebba/esphome/components/foxess_solar/foxess_solar.cpp#L156)

Do you have a log snippet for when it does update the sensors after line 156? Did you capture the logs via the network?

spitfire, 2024-10-20:
> Somehow it's only publishing the sensor states until line [foxess_solar.cpp#L156](https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant/blob/ec5977a3af52e2e402e82bb1fbf97edea416ebba/esphome/components/foxess_solar/foxess_solar.cpp#L156)
> 
> Do you have a log snippet for when it does update the sensors after line 156? Did you capture the logs via the network?
> 
> Can you try this instead of the external_component in your config?
> 
> ```
> external_components:
>   - source: github://SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant@PR
> ```

Updated my config to use PR branch as suggested.

My logs show this so far:

```
[14:59:37][D][sensor:094]: 'WiFi Signal dB': Sending state -50.00000 dBm with 0 decimals of accuracy
[14:59:37][D][sensor:094]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[14:59:44][D][internal_temperature:069]: Ignoring invalid temperature (success=0, value=53.3)
[14:59:46][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[14:59:46][D][esp32.preferences:114]: Saving 1 preferences to flash...
[14:59:46][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:00:37][D][sensor:094]: 'WiFi Signal dB': Sending state -54.00000 dBm with 0 decimals of accuracy
[15:00:37][D][sensor:094]: 'WiFi Signal Percent': Sending state 92.00000 % with 0 decimals of accuracy
[15:00:44][D][internal_temperature:069]: Ignoring invalid temperature (success=0, value=53.3)
[15:00:45][W][FoxessSolar::parse_message:126]: Unexpected msg length, length: 165
[15:00:45][W][component:157]: Component foxess_solar.sensor set Warning flag: unspecified
[15:00:45][D][sensor:094]: 'T-Series Grid Power': Sending state 0.00000 W with 0 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Generation Power': Sending state 1770.00000 W with 0 decimals of accuracy
[15:00:45][D][light:036]: 'Foxess Inverter Light' Setting:
[15:00:45][D][light:047]:   State: ON
[15:00:45][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[15:00:45][D][light:085]:   Transition length: 1.0s
[15:00:45][D][sensor:094]: 'T-Series loads Power': Sending state 0.00000 W with 0 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Grid Voltage R': Sending state 236.00000 V with 1 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Grid Current R': Sending state 2.60000 A with 1 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Grid Frequency R': Sending state 49.96000 Hz with 2 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Grid Power R': Sending state 590.00000 W with 0 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Grid Voltage S': Sending state 232.80000 V with 1 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Grid Current S': Sending state 2.60000 A with 1 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Grid Frequency S': Sending state 49.96000 Hz with 2 decimals of accuracy
[15:00:45][D][sensor:094]: 'T-Series Grid Power S': Sending state 590.00000 W with 0 decimals of accuracy
[15:00:45][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:00:45][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
```

The` [FoxessSolar::parse_message:126]: Unexpected msg length, length: 165` who's that it failed to parse the message (which the implementation you forked from has no issues with), which is why it doesn't show some of the properties.

SibrenVasse, 2024-10-20:
It's just a message to warn that the length of the messages received by the inverter are valid (integrity checked by checksum, which the original implementation does not do), but not the same length as the messages I verified my code with. [here](https://github.com/SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant/blob/ec5977a3af52e2e402e82bb1fbf97edea416ebba/esphome/components/foxess_solar/foxess_solar.cpp#L133)
It will not abort parsing the message, just print the warning to the logs.

The parsing itself has not changed from the original code.

SibrenVasse, 2024-10-20:
I have created a version without this check:

```
external_components:
  - source: github://SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant@PR-nocheck
```

spitfire, 2024-10-20:
> It's just a message to warn that the length of the messages received by the inverter are valid (integrity checked by checksum, which the original implementation does not do), but not the same length [...]

Ok. Something has though as it seems to cut off data received from the inverter with your code, cause it works with the original. Anything else I can check?

spitfire, 2024-10-20:
> I have created a version without this check:
> 
> ```
> external_components:
>   - source: github://SibrenVasse/Foxess-T-series-ESPHome-Home-Assistant@PR-nocheck
> ```

no more warnings, but still just half of the data it seems:
```

[15:14:45][D][sensor:094]: 'T-Series Grid Power': Sending state 0.00000 W with 0 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Generation Power': Sending state 1766.00000 W with 0 decimals of accuracy
[15:14:45][D][light:036]: 'Foxess Inverter Light' Setting:
[15:14:45][D][light:047]:   State: ON
[15:14:45][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[15:14:45][D][light:085]:   Transition length: 1.0s
[15:14:45][D][sensor:094]: 'T-Series loads Power': Sending state 0.00000 W with 0 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Voltage R': Sending state 236.30000 V with 1 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Current R': Sending state 2.60000 A with 1 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Frequency R': Sending state 49.99000 Hz with 2 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Power R': Sending state 588.00000 W with 0 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Voltage S': Sending state 231.80000 V with 1 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Current S': Sending state 2.60000 A with 1 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Frequency S': Sending state 49.99000 Hz with 2 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Power S': Sending state 588.00000 W with 0 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Voltage T': Sending state 240.80000 V with 1 decimals of accuracy
[15:14:45][D][sensor:094]: 'T-Series Grid Current T': Sending state 2.60000 A with 1 decimals of accuracy
[15:15:02][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:15:02][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[15:15:07][D][internal_temperature:069]: Ignoring invalid temperature (success=0, value=53.3)
[15:15:17][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[15:15:17][D][esp32.preferences:114]: Saving 1 preferences to flash...
[15:15:17][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[15:15:17][D][sensor:094]: 'WiFi Signal dB': Sending state -50.00000 dBm with 0 decimals of accuracy
[15:15:17][D][sensor:094]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[15:16:07][D][internal_temperature:069]: Ignoring invalid temperature (success=0, value=53.3)
[15:16:17][D][sensor:094]: 'WiFi Signal dB': Sending state -50.00000 dBm with 0 decimals of accuracy
[15:16:17][D][sensor:094]: 'WiFi Signal Percent': Sending state 100.00000 % with 0 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Power': Sending state 0.00000 W with 0 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Generation Power': Sending state 1945.00000 W with 0 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series loads Power': Sending state 0.00000 W with 0 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Voltage R': Sending state 238.20000 V with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Current R': Sending state 2.80000 A with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Frequency R': Sending state 49.99000 Hz with 2 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Power R': Sending state 648.00000 W with 0 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Voltage S': Sending state 234.30000 V with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Current S': Sending state 2.80000 A with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Frequency S': Sending state 49.99000 Hz with 2 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Power S': Sending state 648.00000 W with 0 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Voltage T': Sending state 240.40001 V with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Current T': Sending state 2.80000 A with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Frequency T': Sending state 49.99000 Hz with 2 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series Grid Power T': Sending state 648.00000 W with 0 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series PV1 Voltage': Sending state 370.39999 V with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series PV1 Current': Sending state 0.90000 A with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series PV1 Power': Sending state 333.35999 W with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series PV2 Voltage': Sending state 366.60001 V with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series PV2 Current': Sending state 4.30000 A with 1 decimals of accuracy
[15:16:45][D][sensor:094]: 'T-Series PV2 Power': Sending state 1576.38000 W with 1 decimals of accuracy
```

spitfire, 2024-10-20:
I don't know how much you have changed, or if it matters but when I compiled the original version with ESP-IDF 5.3.1 it was erroring out with `src/foxess_t_series.h:351:30: error: format '%i' expects argument of type 'int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]`. I know it is just a warning treated as error, but you must have changed something here (validating/converting types?) since your code does compile with it.

SibrenVasse, 2024-10-20:
I haven't tried compiling with esp-idf because of incompatibilities with my hardware.

Can you try compiling with the arduino framework?

spitfire, 2024-10-20:
My inverter is already down for the day. I’m not sure if I will have time to do it tomorrow.

giedriuskilcauskas, 2025-02-22:
ESPHome 2025.2 already removed custom component :(

spitfire, 2025-03-05:
> I haven't tried compiling with esp-idf because of incompatibilities with my hardware.
> 
> Can you try compiling with the arduino framework and enabling verbose logging?

Seems to work fine with Arduino framework.

SibrenVasse, 2025-03-05:
Does it work with esp-idf with the current version of esphome?

aodbq1, 2025-03-10:
Here are my logs:

> INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/foto.yaml...
INFO Starting log output from 192.168.2.140 using esphome API
INFO Successfully connected to foto @ 192.168.2.140 in 0.056s
INFO Successful handshake with foto @ 192.168.2.140 in 0.098s
[10:07:17][W][FoxessSolar::parse_message:134]: Unexpected msg length, length: 245
[10:07:17][W][component:157]: Component foxess_solar.sensor set Warning flag: unspecified
[10:07:52][W][api.connection:093]: : Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[10:08:47][W][component:170]: Component foxess_solar.sensor cleared Warning flag

In my code, I have the source:
 `- source:
    type: local
    path: "components"`

I added 4 files to the folder /components/foxess_solar:
__init__.py
foxess_solar.cpp
foxess_solar.h
sensor.py

What am I doing wrong?
"Previously, I used [this](https://github.com/assembly12/Foxess-T-series-ESPHome-Home-Assistant), and everything worked for several years, but now I have to use external_component

spitfire, 2025-03-13:
> Does it work with esp-idf with the current version of esphome?

I checked it and it does now.

gaszu, 2025-04-05:
spitfire - skąd wziąźć twoje templaty?

eckozor, 2025-12-23:
Hi,
I tried to order pcb from JLCPCB, but after uploading gerber file some of connections are lost. So please if you can send original pcb file. Contact mail: edhem.vikalo@gmail.com
Thank you!

dimweb1-rgb, 2025-12-23:
Hello @eckozor,

I placed an order with jlcpcb three weeks ago and the delivery arrived yesterday. 

I have just tested it and the Gerber file can be uploaded without any problems.

Please try again after clearing your browser's cache.

spitfire, 2025-12-23:
> > [@dimweb1-rgb](https://github.com/dimweb1-rgb): what MAX485 module are you using?
> > 
> > You can enable debug logging to see if the module receives anything over uart at all: (add the debug: line)
> > 
> > ```
> > uart:
> >   id: uart_bus
> >   tx_pin: GPIO16
> >   rx_pin: GPIO17
> >   baud_rate: 9600
> >   debug:
> > ```
> > 
> > if the esp module receives anything at all via uart, you should see the following in the log:
> > 
> > [13:06:55][D][uart_debug:114]: <<< 7E:7E:01:MSG_CONTENT:E7:E7
> > 
> > If you don't receive anything, you could take a look at the flow_control_pin (that defaults to GPIO4). you can find it in the 'configuration variables' section of the readme.
> > 
> > the MAX485 module needs the pin to be correctly configured
> > 
> > I have a Foxess F3000. You can decode the software version from the 0x01 function code message I listed above
> > 
> > ```
> > Master: 1.51
> > Slave: 1.05
> > Manager: 2.30
> > ```
> 
> Hello @SibrenVasse ,
> 
> The PCB board has been soldered, but there is still no data coming from the inverter.
> 
> Can you tell anything from the log?
> 
> [logs_foxess-inverter_logs.txt](https://github.com/user-attachments/files/24314859/logs_foxess-inverter_logs.txt)

It doesn't seem to be related to the issue, can you open another one instead? I'll close this one, since external_component is working, thanks @SibrenVasse !