assembly12/Foxess-T-series-ESPHome-Home-Assistant #25: Not getting any readout

martinjuhasz, 2024-07-07:
Hey,

i've set everything up, but don't get any readout from UART.

I think it might be related to the foxess firmware version.
what is weird is, that i dont have a foxess wifi dongle but an "anatel" branded one.

anyway, when accessing this webinterface i see the following information:

[image]

i tried upgrading to 1.20 using a .bin file i found here in a thread, but it will fail with this message:

[image]

Did anyone encounter this? Any idea how i can fix this or what even might be my problem?

Greatful for any help! I'm really at my wits end here.

assembly12, 2024-07-07:
Mine is from Anatel as well.
Are you able to connect to the foxess cloud (via the foxess app)?

martinjuhasz, 2024-07-07:
@assembly12 yes i am. works fine, connects to my wifi etc.

i run on 1.03 is that correct. how do i update the software version? might this possibly be the problem?

where the heck can i find any of this software or firmware to download? i literrally only found 1.20 here in a thread which is 300kb and will never upload

assembly12, 2024-07-08:
Ok. That's good. Are you sure 1.3 is the firmware of your inverter and not the firmware of your wifi dongle?
I don't believe I've seen inverterfirmware 1.3 come up before.

martinjuhasz, 2024-07-08:
@assembly12 no im not. i thought "hardware version" would indicate that. but no? 

how do i verify which firmware version is running?

assembly12, 2024-07-08:
I believe you can navigate in the menu directly on the inverter and use the display on the inverter. Or look in the App. Im not 100% sure because I no longer use the official App.

martinjuhasz, 2024-07-08:
ah ok i see. on the inverter itself.

i'm running 1.23/1.00/1.14

So i need to downgrade? anyway to do that?

assembly12, 2024-07-08:
I myself have never changed my firmware so can't really help with that. Over at the [home assistant forum](https://community.home-assistant.io/t/foxess-inverter-modbus/459313/174?u=assembly) there is quite some info on this though. I would read up on how to do this over there.

martinjuhasz, 2024-07-08:
sadly not any info their either about downgrading. some seem to be done it successfully, but no mention of how and even where to get the files from. i've added a reply, hopefully someone will be able to help me out. 

thats really a bummer.